fpistm/Arduino_Core_STM32L4 #2: Wire library not working?

evandavey, 2016-10-17:
Wire.endTransmission() always returns 4 (unknown error) on a stock NUCLEO-L476RG.

rogerclarkmelbourne, 2016-10-17:
Thanks.

I'll pass this on to the developers (Wi6Labs)

fprwi6labs, 2016-10-18:
evandavey please check if:
- external pullup resistors are connected to SDA and SCL
- external pullup are connected to 3.3V (not 5V)
- there is no error on communication

evandavey, 2016-10-18:
@fprwi6labs I have tried running I2CScanner code (http://playground.arduino.cc/Main/I2cScanner) both without anything connected (error code 4 on every address) and with a Atmel AT24C256 EEPROM connected along with 10k pull-ups to 3.3V (same results).

The same code and hardware generates no errors and finds the AT24C256 at 0x50 on stock Arduinos (I've tried a Leonardo and Mega2560).

fprwi6labs, 2016-10-19:
Sorry @evandavey but the sketch can't work with Nucleo as it is. The correction is:

`Wire.beginTransmission(address);`
`Wire.write(byte(0x00));`
`error = Wire.endTransmission();`

This is a HAL limitation in your case. You can't send only the address alone. See the function "HAL_I2C_Master_Transmit()" in stm32l4xx_hal_i2c.c for more information.

Furthermore, the unknow error (4) will always be returned because the HAL I2C flags can't match with the Arduino I2C flags 2 & 3 (NACK error).

evandavey, 2016-10-19:
OK, thanks.  Let me fire up some actual code to read/write to the EEPROM and report back.

Are you intending to produce documentation on the differences between standard Arduino?

evandavey, 2016-10-19:
I can confirm that the modifed I2CScanner code finds the EEPROM

```
Scanning...
I2C device found at address 0x50  !
done
```

```c++

#include <Wire.h>
  
void setup()
{
  Wire.begin();
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    Wire.write(byte(0x00)); //Added for NUCLEO
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }

  // removed error code = 4 
   
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(10000);           // wait 5 seconds for next scan
}
```

closed 2016-10-19 by evandavey