fpistm/Arduino_Core_STM32L4 #4: Winterrupts.c writes but never reads .configured

rogerclarkmelbourne, 2016-11-20:
In various lines Winterrupts.c e.g. 

https://github.com/stm32duino/Arduino_Core_STM32L4/blob/master/cores/arduino/WInterrupts.c#L43

.configured is written but never read.

These lines can be removed and 

`PinDescription g_intPinConfigured[MAX_DIGITAL_IOS];`

can be changed to a array of pointers to PinDescription 

(and the code changed accordingly)

fprwi6labs, 2017-01-10:
g_intPinConfigured can be changed to a pointer. 
However, I think .configured should be kept in the code because it is read in other file.
In this case, at line 85, true must be replaced with false.
https://github.com/stm32duino/Arduino_Core_STM32L4/blob/master/cores/arduino/WInterrupts.c#L85

I think the parameter "configured" is not well used in the entire code. I don't know if this parameter is really useful. For the moment we keep it.

rogerclarkmelbourne, 2017-01-10:
We have made a lot of changes to the F1 which probably include changes to this.

The complicated pin mapping function have been simplified etc

Ideally the code for the different MCU series would not differ too much, and perhaps the L4 code should be updated to be more like the revised F1 code.

fpistm, 2017-06-02:
This issue was moved to stm32duino/Arduino_Core_STM32#6

closed 2017-06-02 by fpistm