fpistm/Arduino_Core_STM32L4 #5: Most of the g_analog_config does not need to be in RAM

rogerclarkmelbourne, 2016-11-20:
Only the  `.dacInstance` and `.timInstance` members of g_analog_config need to be in RAM

https://github.com/stm32duino/Arduino_Core_STM32L4/blob/master/system/libstm32l4/source/analog.c#L95

The rest of the struct is const. Hence these 2 members should be in their own separate arrays
g_analog_config should be declared as const
Code should be modified accordingly to match these changes

fprwi6labs, 2017-01-10:
We are currently looking for an alternative to this structure (less code usage and more flexible to integrate variant board).

rogerclarkmelbourne, 2017-01-10:
We have already made changes to the F1 repo to fix this, but I don't know if those changes would be applicable to the L4. 
Its not so much of an issue on the L4 because it has a lot of RAM, but ideally the implementation of this would not differ very much between each MCU series

fpistm, 2017-06-02:
No more applicable on new repo:
[](https://github.com/stm32duino/Arduino_Core_STM32)

closed 2017-06-02 by fpistm