fpistm/Arduino_Core_STM32L4 #7: RNG and other functions are not available

rogerclarkmelbourne, 2016-11-20:
See http://stm32duino.com/viewtopic.php?f=3&t=1569&p=20222#p20222

RNG and possibly other features should be enabled / compiled into the static lib if they don't increase RAM usage

e.g.

https://github.com/stm32duino/Arduino_Core_STM32L4/blob/d2379b2bfe9c685b5dd814fa6284c7fc7facdd5e/system/libstm32l4/include/stm32l4xx_hal_conf.h#L83

danieleff, 2016-11-21:
My first test shows that all modules can be enabled, the linker will remove the unused code, no increase in final flash / ram size. (The test was actually done with F1, not L4, but it should be the same)

rogerclarkmelbourne, 2016-11-21:
@danieleff 

Thanks.

In that case we should enable as much as possible

fpistm, 2017-06-02:
Hi @rogerclarkmelbourne 
on new repo, there is no more static lib. Enable or disable features could be done by updating the desired stm32Yxxx_hal_conf.h in the variant.

closed 2017-06-02 by fpistm