fpistm/Arduino_Core_STM32L4 #8: HAL callback functions in the core should should be declared as weak so they can be overridden in the sketch

rogerclarkmelbourne, 2016-11-23:
HAL callback functions in the core should should be declared as weak so they can be overridden in the sketch

fprwi6labs, 2017-01-10:
Can you declare the same function weak more than one time? 
In the doubt I tried and I can say that it is not possible because gcc returns an error (multiple definition).

rogerclarkmelbourne, 2017-01-10:
OK.
Perhaps there need to be some code changes so they are only declared once as weak.

I'm sure someone else posted about this to the forum, I'll need to find the post and link it here.

fprwi6labs, 2017-01-16:
I'm really curious to know how the linker manages the case where the same function is declared twice as weak. Maybe there is a way but I don't know it...

fpistm, 2017-06-02:
This issue was moved to stm32duino/Arduino_Core_STM32#7

closed 2017-06-02 by fpistm